foxBMS/foxbms-2 #1: System.h file is missing, hardware zip package cannot unzip

XDX2511, 2021-06-01:
This is a very good project, ,but where I can find the System.h file ,and seems hardware zip package demaged cannot unzip

foxBMS, 2021-06-01:
Dear @XDX2511,

generally, you can find the latest version of the documentation rendered online at https://docs.foxbms.org/.

- The hardware zip can be extracted on Windows with common unzipping programs. The PDF is exported by Altium and zipped with 7zip.
- There is no ``System.h`` file in foxBMS 2 project. Information on how to configure foxBMS 2 is found at https://docs.foxbms.org/software/configuration/configuration.html.

Best regards,
The foxBMS Team

closed 2021-06-01 by foxBMS

XDX2511, 2021-06-02:
Dear @foxBMS  ,
      Thank you for your reply , the reason that hardware zip package demaged cannot unzip is I download the file from the github web page ,I use the github desktop no problem .
      One more a question ,where I can get the PCB projec file and headfile HL_system.h ? 

 Looking forward the new SW release and response!

foxBMS, 2021-06-07:
Dear @XDX2511,

Happy to hear that the zip file is now working for you. The PCB will be added in one of the upcoming releases.

The ``HL_system.h`` file is created throught HALCoGen during the build process. You can find it in ``build/bin/src/hal/include``.

Best regards,
The foxBMS Team