foxBMS/foxbms-2 #2: Cell Voltage Transmission Code Fix

colookie, 2021-08-05:
Hello, there is an issue in lines 76 & 77 of the can_cbs_tx_voltage.c file, particularly regarding the start bits. Cell voltage 2 has the start bit for cell voltage 3 and cell voltage 3 overwrites cell voltage 0. The correct start bits to correctly line up with the dbc file are 11, 30, 33, 52.

foxBMS, 2021-08-06:
Dear @colookie,

Thanks for pointing us to this.

Your are absolutely correct, the current implementation is wrong and your solution is correct.
We have fixed this already in our development branch and the fix will be available in the next release.

Best regards,
The foxBMS Team

foxBMS, 2021-08-06:
Dear @colookie,

we have published [v1.1.1](https://github.com/foxBMS/foxbms-2/releases/tag/v1.1.1) that includes the fix for this bug.

Best regards,
The foxBMS Team

closed 2021-08-06 by foxBMS