foxBMS/foxbms-2 #8: Wrong label association to the pins in TI TMS570LC4357 Master v1.1.1 schematics

Shrisha-Balkur, 2022-06-02:
I have two observations: 

1) Based on the datasheet of IC PCF2131TF, pin 12 is BAR(INTB), and pin 13 is BAR(INTA). But in the schematic 'rtc.SchDoc' present in the [TI TMS570LC4357 Master v1.1.1](https://iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/hardware/release/master-v1.1.1.zip), the label assignment is interchanged.  Is this correct?

[image]

[image]

foxBMS, 2022-06-13:
Dear @Shrisha-Balkur,

Thanks for pointing us to this issue.
Yes, you are right, the two pins ``INTA`` and ``INTB`` are reversed. We will fix this issue in the next hardware release.

Best regards,
The foxBMS Team

foxBMS, 2022-07-29:
Dear @Shrisha-Balkur,

thanks again for pointing us to that issue.
This was indeed wrong and we fixed it in the [v1.4.0 release](https://github.com/foxBMS/foxbms-2/releases/tag/v1.4.0), which includes the new hardware schematics etc. for the BMS Master (BMS Master version v1.1.3).

You can find more information of the changes on the hardware side for the BMS Master v1.1.3 [here](https://iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/docs/html/v1.4.0/hardware/master/ti-tms570lc4357-vx.x.x/ti-tms570lc4357-changelog.html#v1-1-3).

Best regards,
The foxBMS Team

closed 2022-07-29 by foxBMS