foxBMS/foxbms-2 #13: MQTT

BhavayAARTECH, 2023-05-06:
Can we implement MQTT over Ethernet Bus foxBMS-2???

closed 2023-05-23 by BhavayAARTECH

foxBMS, 2023-06-27:
Dear @BhavayAARTECH ,

This requires first a port of both, [FreeRTOS-Plus-TCP](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP) and then on top [coreMQTT](https://github.com/FreeRTOS/coreMQTT), to the foxBMS 2 project.

Best regards,
The foxBMS Team

BhavaySen, 2023-08-21:
Hello Team foxBMS,

One part **MC33FS830A0ES** is used in MASTER Card. This is an OTP IC, I have not gone through all the modules of firmware yet but wanted to know that the programming of this IC is done in the firmware or is it already programmed before assembly??
This A0 part of ASIL D category is used in two different flavor. Program to set the desired configuration is covered in firmware in documentation???

foxBMS, 2023-08-23:
Dear @BhavaySen,

The MC33FS830A0ES needs to be programmed before the PCB is assembled. The details of the MC33FS830A0ES one-time programming  cannot be shared due to NDAs.

For future questions, bug reports and similar, please open each time a specific issue.

Best regards,
The foxBMS Team

BhavaySen, 2023-08-24:
Thanks for the response, 
Sure, I'll create a new issue with new queries.