foxBMS/foxbms-2 #14: unable to get local issue certificate

AnkitKherodiya, 2023-07-04:
[image]

While running batch file got local certicate issye, anything to be configured?

foxBMS, 2023-07-04:
Dear @AnkitKherodiya,

The problem arises from the link check that is performed on the official foxBMS documentation site (https://docs.foxbms.org) when the GUI is, e.g., imported.
We had to issue a new certificate and this certificate is not in distributed in the pip package we have in the conda environment.

One work temporary workaround is to ignore the SSL error that is created from these lines
https://github.com/foxBMS/foxbms-2/blob/9eb6d1c44e63e43e62bbf6983b2e618fb6ad02cc/tools/gui/fgui/__init__.py#L63-L67
to use instead
```diff
diff --git "a/tools/gui/fgui/__init__.py" "b/tools/gui/fgui/__init__.py"
index 6104d4fee..94f96c2dd 100644
--- "a/tools/gui/fgui/__init__.py"
+++ "b/tools/gui/fgui/__init__.py"
@@ -42,7 +42,7 @@
 
 import os
 from pathlib import Path
-
+import ssl
 import httplib2
 
 __version__ = "1.5.1"
@@ -65,6 +65,9 @@ try:
 except TimeoutError:
     resp = []
     resp.append({"status": 401})
+except ssl.SSLCertVerificationError:
+    resp = []
+    resp.append({"status": 401})
 if int(resp[0]["status"]) >= 400:
     FOXBMS_DOC_URL = f"{FOXBMS_DOC_BASE_URL}/foxbms/gen2/docs/html/latest/"
 # use latest if it correct version does not exist
```

We will also ship this as workaround in the next release.

foxBMS, 2023-11-22:
The workaround for this issue has been shipped in [release v1.6.0](https://github.com/foxBMS/foxbms-2/releases/tag/v1.6.0).

Best regards,
The foxBMS Team

closed 2023-11-22 by foxBMS