foxBMS/foxbms-2 #15: Feature: Min Max Algo Change

AnkitKherodiya, 2023-07-10:
Hi,

Currently Min Max logic is implemented on Pack level, Shall we implement it based on Module Level too.

As number of module increases, i think it's better to keep track of module min max too. I found it useful for one of the ESS system.

        pMinMaxAverageValues->minimumCellVoltage_mV[s]      = min;
        pMinMaxAverageValues->nrCellMinimumCellVoltage[s]   = cellNumberMinimum;
        pMinMaxAverageValues->nrModuleMinimumCellVoltage[s] = moduleNumberMinimum;
        pMinMaxAverageValues->maximumCellVoltage_mV[s]      = max;
        pMinMaxAverageValues->nrCellMaximumCellVoltage[s]   = cellNumberMaximum;
        pMinMaxAverageValues->nrModuleMaximumCellVoltage[s] = moduleNumberMaximum;
        pMinMaxAverageValues->validMeasuredCellVoltages[s]  = nrValidCellVoltages;

foxBMS, 2023-08-23:
Dear @AnkitKherodiya,

thanks for this suggestion.

We think that this is a useful feature in some use cases and we are going to implement it sometime in the future.

Best regards,
The foxBMS Team

BhavaySen, 2023-08-24:
Hello @AnkitKherodiya 

Hope you are well!

I would like to connect through mail, if you feel free then please connect with me at bsen@aartechsolonics.com. I would like to discuss more regarding foxBMS-2 as my team is also working on the same.

Thank you.

AnkitKherodiya, 2023-08-26:
Hey,

Contacted you.

AnkitKherodiya, 2023-09-09:
for (j = 0; j < BS_NR_OF_BAT_CELLS_PER_MODULE; j++)
        {
            if ((ltc_cellvoltage.valid_volt[i] & (0x01 << j)) == 0)
            {
            	if((ltc_cellvoltage.voltage[i * (BS_NR_OF_BAT_CELLS_PER_MODULE) + j] != 0))
            	{
					/* Cell voltage is valid -> use this voltage for subsequent calculations */
					nrValidCellVoltages++;
					sum += ltc_cellvoltage.voltage[i * (BS_NR_OF_BAT_CELLS_PER_MODULE) + j];

					// Module Min
					if ( (ltc_cellvoltage.voltage[i * (BS_NR_OF_BAT_CELLS_PER_MODULE) + j] < module_min) && \
						   (ltc_cellvoltage.voltage[i * (BS_NR_OF_BAT_CELLS_PER_MODULE) + j] != 0))
					{
						module_min = ltc_cellvoltage.voltage[i * (BS_NR_OF_BAT_CELLS_PER_MODULE) + j];
					}

					// Module Max
					if (ltc_cellvoltage.voltage[i * (BS_NR_OF_BAT_CELLS_PER_MODULE) + j] > module_max)
					{
						module_max = ltc_cellvoltage.voltage[i * (BS_NR_OF_BAT_CELLS_PER_MODULE) + j];
					}

					// Across Min
					if ( (ltc_cellvoltage.voltage[i * (BS_NR_OF_BAT_CELLS_PER_MODULE) + j] < min) && \
						  (ltc_cellvoltage.voltage[i * (BS_NR_OF_BAT_CELLS_PER_MODULE) + j] != 0))
					{
						min = ltc_cellvoltage.voltage[i * (BS_NR_OF_BAT_CELLS_PER_MODULE) + j];
						module_number_min = i;
						cell_number_min = j;
					}
					// Across Max
					if (ltc_cellvoltage.voltage[i * (BS_NR_OF_BAT_CELLS_PER_MODULE) + j] > max)
					{
						max = ltc_cellvoltage.voltage[i * (BS_NR_OF_BAT_CELLS_PER_MODULE) + j];
						module_number_max = i;
						cell_number_max = j;
					}
            	}
            }
        }
        // Set Module level min and max voltages
        ltc_minmax.voltage_min_module[i] = module_min;
        ltc_minmax.voltage_max_module[i] = module_max;
    }

Above logic might help in determining module level min, max and system level min and maximum

closed 2023-09-09 by AnkitKherodiya