foxBMS/foxbms-2 #16: out of bounds

maddin200, 2023-07-17:
src/app/driver/afe/nxp/mc33775a/nxp_mc33775a.c line 549
Code:
     uint16_t uid[3u]                          = {0};
    ...
    for (uint8_t j = 0u; j <= 3u; j++) {
                n775_state->n775Data.uid[n775_state->currentString][i - 1u] |= ((uint64_t)uid[j]) << (16u * j); // uid[j] out of bounds
            }

foxBMS, 2023-08-23:
Hi @maddin200,

many thanks for the hint, we will have a look.

Best regards,
The foxBMS Team

foxBMS, 2024-03-20:
Hi @maddin200,

We verified the behavior and the next release will include a fix for this.

Thanks for the hint!

Best regards,
The foxBMS Team

foxBMS, 2024-08-09:
Hi @maddin200,

The issue has been fixed in foxBMS 2 v1.7.0.

Thanks again for the hint!

Best regards,
The foxBMS Team

closed 2024-08-09 by foxBMS