foxBMS/foxbms-2 #23: Regarding power sequencing and delay for both the PMIC.

BhavaySen, 2023-10-30:
Please share  the detail of  power sequencing  and delays of phase shift of both the PMIC . 
#foxBMS_master_card #TMS570LC4357

foxBMS, 2024-01-22:
Dear @BhavaySen,

could you please elaborate your question in more detail?

Best regards,
The foxBMS Team

BhavaySen, 2024-01-23:
Hello Team,

As I embark on the foxBMS-2 setup, I am facing challenges in completing the PMIC program (OTP) in alignment with the application requirements. I would like to inform you that I hold a valid NDA signed with NXP and I also have the KITFS85SKTEVM Evaluation board for FS8530A0 PMIC.

I kindly request your support in validating and sharing the necessary OTP files, if feasible.

Company Name: Aartech Solonics Limited, Mandideep.

Thank you for your assistance.

Best Regards,
Bhavay Sen

BhavaySen, 2024-01-23:
> Dear @BhavaySen,
> 
> could you please elaborate your question in more detail?
> 
> Best regards, 
> The foxBMS Team

Hello Team,

As I embark on the foxBMS-2 setup, I am facing challenges in completing the PMIC program (OTP) in alignment with the application requirements. I would like to inform you that I hold a valid NDA signed with NXP and I also have the KITFS85SKTEVM Evaluation board for FS8530A0 PMIC.

I kindly request your support in validating and sharing the necessary OTP files, if feasible.

Thank you for your assistance.

Best Regards,
Bhavay Sen

foxBMS, 2024-02-14:
Dear @BhavaySen,

You can find the OTP files at https://iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/hardware/release/.
Please ensure to read and understand the information in `NOTICE.txt` which is included in the .zip file.

Please note that we provide support only for foxBMS boards manufactured and tested by us.
Please use the Issue tracker for bugs and for any technical questions, please reach out to us through [info@foxbms.org](mailto:info@foxbms.org).

Best regards,
The foxBMS Team

foxBMS, 2024-02-14:
/close