foxBMS/foxbms-2 #30: Power sequencing and delay for both the PMIC.

Arrjun8005, 2023-12-19:
Hi , 
I'm receiving an error with Fox BMS code that says ABIST2 fail and BAD_WD_TIMING (wrong window, timeout, or bad WD refresh).
Please find the fail and safe register value after the FS85_InitializeFsPhase for reference.

I have not made any modifications to the code; I am using the same version of fox BMS2.

**- grl_flags	0xE5
- iOvervoltageUndervoltageSafeReaction1	0x9C30
- iOvervoltageUndervoltageSafeReaction2	0x44E2
- iWatchdogConfiguration	0x8120
- i_safe_inputs	0x58E1
- iFailSafeSateMachine	0x5200
- i_svs	0x318
- watchdogWindow	0xBD80
- watchdogSeed	0x8E5
- watchdogAnswer	0x9CC0
- overvoltageUndervoltageRegisterStatus	0x58E2
- releaseFs0bPin	0x8110
- safeIos	0x30E1
- diag_safety	0x5100
- intb_mask	0xC08
- states	0xBD80**

**Here, the configured value in FS85_InitializeFsPhase and the register value are not matching.**

[image]

Best regards,
Arjun.

BhavaySen, 2024-01-23:
Hi @Arrjun8005,

My name is Bhavay, and I am reaching out to you to discuss further details about the PMIC in foxBMS-2. I would appreciate it if you could kindly email me at bsen@aartechsolonics.com to arrange a suitable time for our discussion

Best Regards,
Bhavay Sen

foxBMS, 2024-02-14:
Dear @BhavaySen,

You can find the OTP files at https://iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/hardware/release/.
Please ensure to read and understand the information in `NOTICE.txt` which is included in the .zip file.

Please note that we provide support only for foxBMS boards manufactured and tested by us.
Please use the Issue tracker for bugs and for any technical questions, please reach out to us through [info@foxbms.org](mailto:info@foxbms.org).

Best regards,
The foxBMS Team

closed 2024-02-14 by foxBMS