foxBMS/foxbms-2 #31: Facing Issue while re-installation

KumarPravaig, 2023-12-25:
Iam facing an error with the software installation of the "foxbms", I followed the steps for installation in the given sequence and it was working good in the past. Now it just started to pop an error in the building process as below:

      ""C:\Users\Kumar\Documents\foxbms-2>tools\utils\cmd\run-python-script.bat tests\env\packages_test.py -f tests\env\conda_env_win32.json Traceback (most recent call last):   File "C:\Users\Kumar\Documents\foxbms-2\tests\env\packages_test.py", line 51, in <module>     import pathlib ModuleNotFoundError: No module named 'pathlib'""

                  I tried many ways solving the problem. But the end result was the same, struggling for the "pathlib" library to recognize. It couldn't build the project.
[image]

foxBMS, 2024-01-22:
Dear @KumarPravaig,

`pathlib` is part of th [Python standard library](https://docs.python.org/3/library/pathlib.html), therefore this error is rather strange.

- Where is your conda environment `2023-02-fennec-fox` install into?
- Can you activate the environment `2023-02-fennec-fox` and successfully run `python -c "import pathlib"`?

Best regards,
The foxBMS Team

foxBMS, 2024-02-14:
Dear @KumarPravaig,

are there any updates on your side?

Best regards,
The foxBMS Team