foxBMS/foxbms-2 #34: SBCS does not power the MCU

taijisugiyama, 2024-01-22:
Greetings FoxBMS team,

So I got foxbms master board manufactured from eurocircuit. I also got the debugger and setup the softwares and environment. Today, I powered the master but the MCU 3.3V or the PER 5V LED does not light up. I have diagnosed with DMM that the SBCS does get the 12V from VSUP 30 but somehow does not output the 3.3V required to run the MCU. As such the master is not working for me. I would be extremely grateful if you can help me find what went wrong. Thank you so much!

taijisugiyama, 2024-01-22:
I am using a 12V 1A power adapter to power the Master

BhavaySen, 2024-01-23:
> Greetings FoxBMS team,
>
> So I got foxbms master board manufactured from eurocircuit. I also got the debugger and setup the softwares and environment. Today, I powered the master but the MCU 3.3V or [...]

@taijisugiyama Have you completed the OTP configurations for the PMIC used in the master card?

foxBMS, 2024-01-23:
Hi @taijisugiyama,

happy to hear that you got the BMS-Master manufactured.

Each of both SBCs needs to be OTP-programmed with the specific correct setting, so that it can supply the MCU and the peripherals etc. with the correct voltage as it is desgined (see the Altium files for that).
The how-to of programming the SBC cannot be shared public (see https://github.com/foxBMS/foxbms-2/issues/20#issuecomment-1691334640, you have to get yourself in contact with the manufacturer yourself to get this information) and OTP the SBC yourself.

Best regards,
The foxBMS Team

taijisugiyama, 2024-01-24:
Thank you so much for the reply!

I am trying to get the OTP for the SBC chip. I have found the evaluation board KITFS85SKTEVM from NXP to program the chip. The problem is that the chips are already soldered to the master. Is there a way to OTP the chip without having to desolder and remove the chip from master? I have noticed we have J3000/J3001 which access the I2C and SPI of the chip along with its debug pins. For what purpose do we have these terminals and can I use them to OTP the chips on board? Thank you so much for the answers, you have helped me tremendously.

foxBMS, 2024-02-14:
Dear @BhavaySen,

You can find the OTP files at https://iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/hardware/release/.
Please ensure to read and understand the information in `NOTICE.txt` which is included in the .zip file.

Please note that we provide support only for foxBMS boards manufactured and tested by us.
Please use the Issue tracker for bugs and for any technical questions, please reach out to us through [info@foxbms.org](mailto:info@foxbms.org).

Best regards,
The foxBMS Team

closed 2024-02-14 by foxBMS