foxBMS/foxbms-2 #36: Issue with foxBMS-2 Master Board Power-Up

BhavaySen, 2024-02-13:
Hello Team foxBMS-2,

I wanted to inform you that we have successfully manufactured the master card and programmed the PMIC with the OTP, thanks to the efforts of the team.

However, during testing, we encountered an issue. When testing the PMIC in Emulation mode, it performed as expected, and we verified the voltages on each PMIC (MCU and PER). However, upon soldering them onto the MASTER Board after the OTP and powering up the circuit with a 12V Supply at connector J2009, we observed that the board did not power up. Additionally, the LEDs indicating different voltages on the MCU_SUP PMIC were blinking at a constant frequency.

Upon further investigation, I discovered that the PMIC requires consecutive WD refreshes to start all the regulators. However, my on-board processor is not programmed, and it cannot be programmed without power, which is to be delivered from the MCU_SUP PMIC.

I have tried using the J-Link Plus, but it cannot deliver power to the board. I also confirmed that the Mictor-38 connector takes power from the MCU_PER PMIC VPRE (3.3V).

I kindly request your assistance in resolving this issue.

Best Regards,
Bhavay Sen

foxBMS, 2024-02-14:
Dear @BhavaySen,

You have set the SBC into debug mode via R7219 or S7200B (see the schematics for details).
You cannot drive the foxBMS-Master via the Mictor-38 connector, you have to supply it as described in the [documentation](https://docs.foxbms.org/hardware/master/ti-tms570lc4357-vx.x.x/ti-tms570lc4357-v1.1.1.html#master-tms570-v1-1-1).

Best regards,
The foxBMS Team

closed 2024-02-14 by foxBMS

BhavaySen, 2024-02-15:
> S7200B

Thanks for the reply, Team foxBMS. However, I noticed that the [documentation](https://docs.foxbms.org/hardware/master/ti-tms570lc4357-vx.x.x/ti-tms570lc4357-v1.1.1.html#master-tms570-v1-1-1) you attached (documentation) does not mention anything about the Ignition signal. I found a section in the documentation ([SBC SECTION 4.34](https://docs.foxbms.org/software/modules/driver/sbc/sbc.html#module-files)) which mentions the requirement for an Ignition signal, but it does not specify the rating of the signal. I believe this information should be included in the Electrical Connection section.

I am looking forward to your response.

Best Regards,
Bhavay Sen

foxBMS, 2024-02-19:
Dear @BhavaySen,

We will update the documentation with the supply range of the ignition signal.
We will also update the documentation regarding the second part of the question you mentioned in #36 (SBC, initial programming and startup).

Best regards,
The foxBMS Team

BhavaySen, 2024-02-20:
> Dear @BhavaySen,
> 
> We will update the documentation with the supply range of the ignition signal. We will also update the documentation regarding the second part of the question you mentioned in #36 (SBC, initial programming and startup).
> 
> Best regards, The foxBMS Team

Thank you team for considering my request.
Please update the mentioned info ASAP. Thanks.

With the same, I have one more doubt regarding the shared OTP file https://iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/hardware/release/ in this OTP file, at the OTP REGISTER 
Fail safe - OTP_CFG_I2C Address 0x15
Main - OTP_CFG_I2C Address 0x2B 
These both have addressed as 0000 in both. As per the datasheet, address for each block should be different. The same case is with MCU_PER IC as well, so there are total two PMICs with same address and in which their two individual block (Main and Fail-safe) also have the same I2C device Address.

Please resolve this query.

Thanks & Regards
Bhavay