foxBMS/foxbms-2 #38: gui.bat problem

BhavaySen, 2024-03-14:
while running the GUI.bat after complete setup and successful build. Below is the output showing on CMD instead of the GUI pops up.

________________________________________________________
D:\foxBMS\foxbms-2>tools\gui\gui.bat
DEBUG:root:Logging level: DEBUG
INFO:root:Checking for pcan in ['PCAN_USBBUS1', 'PCAN_USBBUS2', 'PCAN_USBBUS3', 'PCAN_USBBUS4', 'PCAN_USBBUS5', 'PCAN_USBBUS6', 'PCAN_USBBUS7', 'PCAN_USBBUS8', 'PCAN_USBBUS9', 'PCAN_USBBUS10', 'PCAN_USBBUS11', 'PCAN_USBBUS12', 'PCAN_USBBUS13', 'PCAN_USBBUS14', 'PCAN_USBBUS15', 'PCAN_USBBUS16']
DEBUG:root:Checking PCAN_USBBUS1 (0)
DEBUG:can:can config: {'state': <BusState.ACTIVE: 1>, 'channel': 'PCAN_USBBUS1', 'interface': 'pcan'}
DEBUG:root:Checking PCAN_USBBUS2 (1)
DEBUG:can:can config: {'state': <BusState.ACTIVE: 1>, 'channel': 'PCAN_USBBUS2', 'interface': 'pcan'}
DEBUG:root:Checking PCAN_USBBUS3 (2)
DEBUG:can:can config: {'state': <BusState.ACTIVE: 1>, 'channel': 'PCAN_USBBUS3', 'interface': 'pcan'}
DEBUG:root:Checking PCAN_USBBUS4 (3)
DEBUG:can:can config: {'state': <BusState.ACTIVE: 1>, 'channel': 'PCAN_USBBUS4', 'interface': 'pcan'}
DEBUG:root:Checking PCAN_USBBUS5 (4)
DEBUG:can:can config: {'state': <BusState.ACTIVE: 1>, 'channel': 'PCAN_USBBUS5', 'interface': 'pcan'}
DEBUG:root:Checking PCAN_USBBUS6 (5)
DEBUG:can:can config: {'state': <BusState.ACTIVE: 1>, 'channel': 'PCAN_USBBUS6', 'interface': 'pcan'}
DEBUG:root:Checking PCAN_USBBUS7 (6)
DEBUG:can:can config: {'state': <BusState.ACTIVE: 1>, 'channel': 'PCAN_USBBUS7', 'interface': 'pcan'}
DEBUG:root:Checking PCAN_USBBUS8 (7)
DEBUG:can:can config: {'state': <BusState.ACTIVE: 1>, 'channel': 'PCAN_USBBUS8', 'interface': 'pcan'}
DEBUG:root:Checking PCAN_USBBUS9 (8)
DEBUG:can:can config: {'state': <BusState.ACTIVE: 1>, 'channel': 'PCAN_USBBUS9', 'interface': 'pcan'}
DEBUG:root:Checking PCAN_USBBUS10 (9)
DEBUG:can:can config: {'state': <BusState.ACTIVE: 1>, 'channel': 'PCAN_USBBUS10', 'interface': 'pcan'}
DEBUG:root:Checking PCAN_USBBUS11 (10)
DEBUG:can:can config: {'state': <BusState.ACTIVE: 1>, 'channel': 'PCAN_USBBUS11', 'interface': 'pcan'}
DEBUG:root:Checking PCAN_USBBUS12 (11)
DEBUG:can:can config: {'state': <BusState.ACTIVE: 1>, 'channel': 'PCAN_USBBUS12', 'interface': 'pcan'}
DEBUG:root:Checking PCAN_USBBUS13 (12)
DEBUG:can:can config: {'state': <BusState.ACTIVE: 1>, 'channel': 'PCAN_USBBUS13', 'interface': 'pcan'}
DEBUG:root:Checking PCAN_USBBUS14 (13)
DEBUG:can:can config: {'state': <BusState.ACTIVE: 1>, 'channel': 'PCAN_USBBUS14', 'interface': 'pcan'}
DEBUG:root:Checking PCAN_USBBUS15 (14)
DEBUG:can:can config: {'state': <BusState.ACTIVE: 1>, 'channel': 'PCAN_USBBUS15', 'interface': 'pcan'}
DEBUG:root:Checking PCAN_USBBUS16 (15)
DEBUG:can:can config: {'state': <BusState.ACTIVE: 1>, 'channel': 'PCAN_USBBUS16', 'interface': 'pcan'}
INFO:root:Connected CAN adapters: [<can.interfaces.pcan.pcan.PcanBus object at 0x000002460E7F3940>]
DEBUG:root:CAN adapter: PCAN_USBBUS1

D:\foxBMS\foxbms-2>
____________________________________________________________

Please check and provide some solution.

Best Regards,
Bhavay Sen

foxBMS, 2024-03-20:
Dear @BhavaySen,

could you please try the following script and provide feedback whether it works or not.

<details>
  <summary>tools/gui/gui.bat</summary>
  
  ```bat
@REM Copyright (c) 2010 - 2024, Fraunhofer-Gesellschaft zur Foerderung der angewandten Forschung e.V.
@REM All rights reserved.
@REM
@REM SPDX-License-Identifier: BSD-3-Clause
@REM
@REM Redistribution and use in source and binary forms, with or without
@REM modification, are permitted provided that the following conditions are met:
@REM
@REM 1. Redistributions of source code must retain the above copyright notice, this
@REM    list of conditions and the following disclaimer.
@REM
@REM 2. Redistributions in binary form must reproduce the above copyright notice,
@REM    this list of conditions and the following disclaimer in the documentation
@REM    and/or other materials provided with the distribution.
@REM
@REM 3. Neither the name of the copyright holder nor the names of its
@REM    contributors may be used to endorse or promote products derived from
@REM    this software without specific prior written permission.
@REM
@REM THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
@REM AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
@REM IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
@REM DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE
@REM FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
@REM DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR
@REM SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
@REM CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY,
@REM OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
@REM OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
@REM
@REM We kindly request you to use one or more of the following phrases to refer to
@REM foxBMS in your hardware, software, documentation or advertising materials:
@REM
@REM - "This product uses parts of foxBMS&reg;"
@REM - "This product includes parts of foxBMS&reg;"
@REM - "This product is derived from foxBMS&reg;"

@SETLOCAL EnableExtensions EnableDelayedExpansion

@TITLE foxBMS 2 GUI

@REM Set the code page to 850
@REM https://docs.microsoft.com/en-us/windows-server/administration/windows-commands/chcp
@REM Get the current code page
@FOR /F "tokens=* USEBACKQ" %%F IN (`chcp.com 2^>^&1`) DO @(
    @SET CODEPAGE_OUT=%%F
)
@REM split output by whitespace and use the last argument as this contains the
@REM activate code page
@FOR %%a in (%CODEPAGE_OUT%) do @(
    @SET CODEPAGE=%%a
)
@REM Remove the trailing dot if it exists
@IF "%CODEPAGE:~-1%"=="." @(
    @SET CODEPAGE=%CODEPAGE:~0,-1%
) ELSE (
    @SET CODEPAGE=%CODEPAGE%
)

@REM If the code page is not multilingual, set it to 850
@IF NOT "%CODEPAGE%"=="850" (
    @chcp.com 850 2>&1 > nul
)
@REM now we have code page that will work
@SET CONDA_BASE_ENVIRONMENT_ACTIVATE_SCRIPT=""
@CALL %~dp0\..\utils\cmd\find_base_conda.bat

@CALL %~dp0\..\utils\cmd\add_pcan_to_path.bat

@IF %CONDA_BASE_ENVIRONMENT_ACTIVATE_SCRIPT%=="" (
    pause
    @EXIT /b 1
)

@CALL %CONDA_BASE_ENVIRONMENT_ACTIVATE_SCRIPT% %CONDA_DEVELOPMENT_ENVIRONMENT_NAME%

@PUSHD %~dp0
@SET PY_EXE=python
@WHERE %PY_EXE% 1>NUL 2>NUL
@IF %ERRORLEVEL% neq 0 SET PY_EXE=py
@%PY_EXE% -m fgui %* -vvvv
@IF %ERRORLEVEL% NEQ 0 (
    @chcp.com %CODEPAGE% 2>&1 > nul
    @EXIT /b %ERRORLEVEL%
) ELSE (
    @chcp.com %CODEPAGE% 2>&1 > nul
    @conda deactivate
    @EXIT /b %ERRORLEVEL%
)
@POPD

  ```
</details>

Best regards,
The foxBMS Team

BhavaySen, 2024-03-21:
> Dear @BhavaySen,
> 
> could you please try the following script and provide feedback whether it works or not.
> 
> tools/gui/gui.bat
> Best regards, The foxBMS Team

I used the same script to run the foxBMS GUI. but still getting the same error.

Best regards,
Bhavay Sen

sankadiaji, 2024-03-31:
still getting the  same issue

foxBMS, 2024-04-16:
Closing this issue in favor of #28 as the debug message the same as shown in https://github.com/foxBMS/foxbms-2/issues/28#issuecomment-1845292877.

Best regards,
The foxBMS Team

closed 2024-04-16 by foxBMS