foxBMS/foxbms-2 #43: PER SBC chip only works in debug mode

taijisugiyama, 2024-04-16:
Greetings,

The PER SBC chip fails Abist1 test when I OTP with the provided script. It only seems to work when I run it in debug mode. Is there any explanation or solution for this?

Thank you!

BhavaySen, 2024-11-08:
> Greetings,
> 
> The PER SBC chip fails Abist1 test when I OTP with the provided script. It only seems to work when I run it in debug mode. Is there any explanation or solution for this?
> 
> Thank you!

@taijisugiyama 
I think, they forgot to add the initialization of PMIC SBC_PER. 
- No SPI initialization found.
- No FS_STATE_INIT found in the code.  

Please share the implementation. if you succeed in initializing the PMIC_PER on master card.

foxBMS, 2024-12-20:
Dear @taijisugiyama,

The power supply architecture has been streamlined in BMS-Master release v1.2.2 and the peripheral SBC became obsolete (see [iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/hardware/release/](https://iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/hardware/release/).

Best regards,
The foxBMS Team

closed 2024-12-20 by foxBMS