foxBMS/foxbms-2 #44: Slave card MC33775A

Prahlad1208, 2024-04-18:
Dear Foxbms Team,
We are applying 14 cell voltages in slave cards.
TPL low signal applied as an input signals through interface MC33664A in Pin no 45 and 46. but **TPL High as an output do not give any output signals in pin no. 47 and 48.**

Regards
Prahlad

Prahlad1208, 2024-06-14:
Dear Foxbms Team,

Pins 47 and 48 provide signals when we apply an external 12 volt supply for Vbat and 5 volt supply for VDDC and VDDIO. 
MC33775A is unable to control the VDDC_DRIVE signal to the external BJT MJD31CAJ, which is required in order to supply 5 volts from pin 42 of MC33775A to the external BJT.

Any solution for VDDC and VDDIO without applying external 5 volt signal.

Best Regards,
Prahlad

foxBMS, 2024-09-26:
Dear @Prahlad1208,

are you using the latest release of the NXP MC33557A-based foxBMS BMS-Slave (version 1.0.0, https://[iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/hardware/release/](https://iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/hardware/release/)).

This BMS-Slave works in combination with NSP MC33664-based BMS-Interface (version 1.0.2, available at the same link).

Which software version of foxBMS 2 are you using?

Best regards,
The foxBMS Team

foxBMS, 2025-04-01:
Dear @Prahlad1208,
 
the MC33775A features an overcurrent threshold on the 5V supply that can be set via resistor R618. When this threshold is exceeded, the MC33775A performs a reset. In the previous hardware versions, this threshold was configured for the minimal population variants which have less power consumption. When populating the complete variant, the overcurrent threshold will trigger a reset resulting in the described behaviour.

In the updated version, the current threshold has been increased to prevent the MC33775A from entering overcurrent mode during full operation.
The resistor R618 has now been reduced to a value of 4.53 Ω and a specific note has been added to the schematic.
 
The updated release can be found at https://iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/hardware/release/slave-14-nxp-mc33775a-v1.0.3.latest.zip.
 
Best regards,
The foxBMS Team

closed 2025-04-01 by foxBMS