foxBMS/foxbms-2 #48: Toolchain installation error

niktwo17, 2024-06-19:
Hi, I have followed the instructions for setting up the environment [here ](https://iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/docs/html/latest/getting-started/software-installation.html), but I am getting strange errors. Happened both with source and latest release:

- The test fails

> PS C:\Users\nremi\Documents\foxbms-2> tools\utils\cmd\run-python-script.bat tests\env\packages_test.py -f tests\env\conda_env_win32.json
Traceback (most recent call last):
  File "C:\Users\nremi\Documents\foxbms-2\tests\env\packages_test.py", line 156, in <module>
    main()
  File "C:\Users\nremi\Documents\foxbms-2\tests\env\packages_test.py", line 96, in main
    raise BaseException("Environments differ.")
BaseException: Environments differ.
PS C:\Users\nremi\Documents\foxbms-2> tools\utils\cmd\run-python-script.bat tests\env\packages_test.py -f tests\env\conda_env_win32.json
Traceback (most recent call last):
  File "C:\Users\nremi\Documents\foxbms-2\tests\env\packages_test.py", line 156, in <module>
    main()
  File "C:\Users\nremi\Documents\foxbms-2\tests\env\packages_test.py", line 96, in main
    raise BaseException("Environments differ.")
BaseException: Environments differ.

- waf configure and ide.bat both fail with a package error

> PS C:\Users\nremi\Documents\foxbms-2> .\waf configure
Waf: The wscript in 'C:\\Users\\nremi\\Documents\\foxbms-2' is unreadable
Traceback (most recent call last):
  File "C:\Users\nremi\Documents\foxbms-2\tools\waf3-2.0.22-1241519b19b496207abef1f72bbf61c2\waflib\Scripting.py", line 102, in waf_entry_point
    set_main_module(wscript)
  File "C:\Users\nremi\Documents\foxbms-2\tools\waf3-2.0.22-1241519b19b496207abef1f72bbf61c2\waflib\Scripting.py", line 142, in set_main_module
    Context.g_module=Context.load_module(file_path)
  File "C:\Users\nremi\Documents\foxbms-2\tools\waf3-2.0.22-1241519b19b496207abef1f72bbf61c2\waflib\Context.py", line 362, in load_module
    exec(compile(code,path,'exec'),module.__dict__)
  File "C:\Users\nremi\Documents\foxbms-2\wscript", line 61, in <module>
    import jsonschema
ModuleNotFoundError: No module named 'jsonschema'`

- manually adding the jsonschema package to the conda environment will just trigger yet another missing package error

Thanks for your support!

foxBMS, 2024-06-19:
Dear @niktwo17,

- Open a cmd.exe 
- Run: `%USERPROFILE%\Miniconda3\Scripts\activate`
- Run: `conda activate 2023-02-fennec-fox`
- Run: `pip install --no-deps alabaster==0.7.12 appdirs==1.4.4 argon2-cffi==21.1.0 argparse-addons==0.7.0 arpeggio==1.10.2 astroid==2.8.4 attrs==21.2.0 babel==2.9.1 backcall==0.2.0 beautifulsoup4==4.10.0 bitstruct==8.11.1 black==21.10b0 bleach==4.1.0 bokeh==2.4.1 bottle==0.12.19 cantools==36.5.0 cffconvert==2.0.0 cffi==1.15.0 charset-normalizer==2.0.7 click==8.0.3 cloudpickle==2.0.0 colorama==0.4.4 coverage==6.1.1 crcmod==1.7 cycler==0.11.0 dask==2021.11.1 debugpy==1.5.1 decorator==5.1.0 defusedxml==0.7.1 diskcache==5.2.1 docopt==0.6.2 docutils==0.16 entrypoints==0.3 et-xmlfile==1.1.0 filterpy==1.4.5 fsspec==2021.11.0 gcovr==5.0 gitdb==4.0.9 gitpython==3.1.24 gprof2dot==2021.2.21 httplib2==0.20.2 idna==3.3 imagesize==1.3.0 ipykernel==6.5.0 ipython==7.29.0 ipython-genutils==0.2.0 ipywidgets==7.6.5 isort==5.10.1 jdcal==1.4.1 jedi==0.18.0 jinja2==2.11.3 joblib==1.1.0 jsonschema==3.2.0 jupyter==1.0.0 jupyter-client==7.0.6 jupyter-console==6.4.0 jupyter-core==4.9.1 jupyterlab-pygments==0.1.2 jupyterlab-widgets==1.0.2 kiwisolver==1.3.2 latexcodec==2.0.1 lazy-object-proxy==1.6.0 locket==0.2.1 lxml==4.6.4 markdown==3.3.4 markupsafe==2.0.1 matplotlib==3.4.3 matplotlib-inline==0.1.3 mccabe==0.6.1 mistune==0.8.4 mypy==0.910 mypy-extensions==0.4.3 nbclient==0.5.5 nbconvert==6.2.0 nbformat==5.1.3 nest-asyncio==1.5.1 notebook==6.4.5 numpy==1.22.4 openpyxl==2.6.4 packaging==21.2 pandas==1.3.4 pandas-stubs==1.2.0.37 pandocfilters==1.5.0 parso==0.8.2 partd==1.2.0 pathspec==0.9.0 patsy==0.5.2 pickleshare==0.7.5 pillow==8.4.0 pip-licenses==3.5.3 plantuml==0.3.0 plantuml-markdown==3.4.4 platformdirs==2.4.0 prometheus-client==0.12.0 prompt-toolkit==3.0.22 psutil==5.8.0 ptable==0.9.2 pybtex==0.24.0 pybtex-docutils==1.0.1 pycparser==2.21 pyenchant==3.2.2 pygments==2.10.0 pykwalify==1.8.0 pylint==2.11.1 pyparsing==2.4.7 pyrsistent==0.18.0 pyserial==3.5 pysimplegui==4.55.1 python-can==4.0.0.dev2 python-datauri==0.2.9 python-dateutil==2.8.2 python-lin==0.1.1 python-markdown-math==0.6 python-uds==1.0.2 pytz==2021.3 pyvisa==1.11.3 pywinpty==1.1.5 pyyaml==5.4.1 pyzmq==22.3.0 qtconsole==5.2.0 qtpy==1.11.2 regex==2021.11.10 requests==2.26.0 requests-file==1.5.1 rope==0.21.0 rsinstrument==1.19.0.75 ruamel-yaml==0.17.17 ruamel-yaml-clib==0.2.6 scipy==1.7.2 seaborn==0.11.2 send2trash==1.8.0 six==1.16.0 smmap==5.0.0 snowballstemmer==2.1.0 soupsieve==2.3 sphinx==4.3.0 sphinx-copybutton==0.3.3 sphinx-panels==0.6.0 sphinx-rtd-theme==1.0.0 sphinxcontrib-applehelp==1.0.2 sphinxcontrib-bibtex==2.4.1 sphinxcontrib-devhelp==1.0.2 sphinxcontrib-htmlhelp==2.0.0 sphinxcontrib-jsmath==1.0.1 sphinxcontrib-mermaid==0.7.1 sphinxcontrib-plantuml==0.21 sphinxcontrib-programoutput==0.17 sphinxcontrib-qthelp==1.0.3 sphinxcontrib-serializinghtml==1.1.5 sphinxcontrib-spelling==7.2.1 statsmodels==0.13.0 tabulate==0.8.9 tenacity==8.0.1 terminado==0.12.1 testpath==0.5.0 textparser==0.23.0 textx==2.3.0 threadpoolctl==3.0.0 toml==0.10.2 tomli==1.2.2 toolz==0.11.2 tornado==6.1 traitlets==5.1.1 typing-extensions==3.10.0.2 uptime==3.0.1 urllib3==1.26.7 wcwidth==0.2.5 webencodings==0.5.1 widgetsnbextension==3.5.2 windows-curses==2.2.0 wrapt==1.13.3 wxpython==4.1.1 xlsxwriter==1.4.5`

Then open a new cmd.exe and try running `waf configure` in the repository.
It should work fine then.

Best regards,
The foxBMS Team

niktwo17, 2024-06-19:
Hi, thanks for your reply - unfortunately the error persists - `waf configure` and `ide.bat` still fail. (all mentioned packages were already present in the environment, nothing was newly installed, except for a version error for pysimplegui):
> ERROR: Could not find a version that satisfies the requirement pysimplegui==4.55.1 (from versions: 2.7.0, 4.60.5, 5.0.0, 5.0.2, 5.0.3, 5.0.4, 5.0.5)
ERROR: No matching distribution found for pysimplegui==4.55.1

I think I found the root cause - apparently legacy versions for pysimplegui were deleted recently, which now breaks the hardcoded v4.55.1 dependency: https://www.reddit.com/r/Python/comments/1d8d4iv/psa_pysimplegui_has_deleted_almost_all_old_lgpl/
I tried using 4.60.5 to replace it, but this breaks some dependency later on:

[image]

foxBMS, 2024-06-19:
Please use
- `waf configure --ignore-env-check` or
- hardcode it into your repository copy somewhere right after the beginning of function `configure` in the top-level `wscript` the following line `conf.options.IGNORE_ENV_CHECK=True`
  ```diff
  diff --git a/wscript b/wscript
  index 181c72c45..f9a885987 100644
  --- a/wscript
  +++ b/wscript
  @@ -311,6 +311,7 @@ def configure(conf):  # pylint: disable=too-many-statements,too-many-branches

       A workspace is generated if Visual Studio Code is found on the machine.
       """
  +    conf.options.IGNORE_ENV_CHECK=True
       if " " in conf.path.abspath():
       conf.fatal(f"Project path must not contain spaces ({conf.path}).")
       conf.env.append_unique("PROJECT_ROOT", pathlib.Path(conf.path.abspath()).as_posix())
  ```

Does it then work for you?

niktwo17, 2024-06-20:
Hi, this will still not work: The `conda-update-env.bat` will not install the environment correctly, the
```
ERROR: Could not find a version that satisfies the requirement pysimplegui==4.55.1 (from versions: 2.7.0, 4.60.5, 5.0.0, 5.0.2, 5.0.3, 5.0.4, 5.0.5)
ERROR: No matching distribution found for pysimplegui==4.55.1
```
causes it to terminate without installing everything (hence the further missing packages such as `jsconschema` etc. I earlier reported

foxBMS, 2024-06-20:
You do not need to run this script. Run the command the provided in https://github.com/foxBMS/foxbms-2/issues/48#issuecomment-2178625454 to install all dependencies in the specific environment instead.

foxBMS, 2024-07-02:
Dear @niktwo17,

there was a copy-paste error in https://github.com/foxBMS/foxbms-2/issues/48#issuecomment-2178625454 (We fixed that also in the comment).

Just to repeat it, the correct commands are:

- Open a cmd.exe
- Run: `%USERPROFILE%\Miniconda3\Scripts\activate`
- Run: `conda activate 2023-02-fennec-fox`
- Run: `pip install --no-deps alabaster==0.7.12 appdirs==1.4.4 argon2-cffi==21.1.0 argparse-addons==0.7.0 arpeggio==1.10.2 astroid==2.8.4 attrs==21.2.0 babel==2.9.1 backcall==0.2.0 beautifulsoup4==4.10.0 bitstruct==8.11.1 black==21.10b0 bleach==4.1.0 bokeh==2.4.1 bottle==0.12.19 cantools==36.5.0 cffconvert==2.0.0 cffi==1.15.0 charset-normalizer==2.0.7 click==8.0.3 cloudpickle==2.0.0 colorama==0.4.4 coverage==6.1.1 crcmod==1.7 cycler==0.11.0 dask==2021.11.1 debugpy==1.5.1 decorator==5.1.0 defusedxml==0.7.1 diskcache==5.2.1 docopt==0.6.2 docutils==0.16 entrypoints==0.3 et-xmlfile==1.1.0 filterpy==1.4.5 fsspec==2021.11.0 gcovr==5.0 gitdb==4.0.9 gitpython==3.1.24 gprof2dot==2021.2.21 httplib2==0.20.2 idna==3.3 imagesize==1.3.0 ipykernel==6.5.0 ipython==7.29.0 ipython-genutils==0.2.0 ipywidgets==7.6.5 isort==5.10.1 jdcal==1.4.1 jedi==0.18.0 jinja2==2.11.3 joblib==1.1.0 jsonschema==3.2.0 jupyter==1.0.0 jupyter-client==7.0.6 jupyter-console==6.4.0 jupyter-core==4.9.1 jupyterlab-pygments==0.1.2 jupyterlab-widgets==1.0.2 kiwisolver==1.3.2 latexcodec==2.0.1 lazy-object-proxy==1.6.0 locket==0.2.1 lxml==4.6.4 markdown==3.3.4 markupsafe==2.0.1 matplotlib==3.4.3 matplotlib-inline==0.1.3 mccabe==0.6.1 mistune==0.8.4 mypy==0.910 mypy-extensions==0.4.3 nbclient==0.5.5 nbconvert==6.2.0 nbformat==5.1.3 nest-asyncio==1.5.1 notebook==6.4.5 numpy==1.22.4 openpyxl==2.6.4 packaging==21.2 pandas==1.3.4 pandas-stubs==1.2.0.37 pandocfilters==1.5.0 parso==0.8.2 partd==1.2.0 pathspec==0.9.0 patsy==0.5.2 pickleshare==0.7.5 pillow==8.4.0 pip-licenses==3.5.3 plantuml==0.3.0 plantuml-markdown==3.4.4 platformdirs==2.4.0 prometheus-client==0.12.0 prompt-toolkit==3.0.22 psutil==5.8.0 ptable==0.9.2 pybtex==0.24.0 pybtex-docutils==1.0.1 pycparser==2.21 pyenchant==3.2.2 pygments==2.10.0 pykwalify==1.8.0 pylint==2.11.1 pyparsing==2.4.7 pyrsistent==0.18.0 pyserial==3.5 python-can==4.0.0.dev2 python-datauri==0.2.9 python-dateutil==2.8.2 python-lin==0.1.1 python-markdown-math==0.6 python-uds==1.0.2 pytz==2021.3 pyvisa==1.11.3 pywinpty==1.1.5 pyyaml==5.4.1 pyzmq==22.3.0 qtconsole==5.2.0 qtpy==1.11.2 regex==2021.11.10 requests==2.26.0 requests-file==1.5.1 rope==0.21.0 rsinstrument==1.19.0.75 ruamel-yaml==0.17.17 ruamel-yaml-clib==0.2.6 scipy==1.7.2 seaborn==0.11.2 send2trash==1.8.0 six==1.16.0 smmap==5.0.0 snowballstemmer==2.1.0 soupsieve==2.3 sphinx==4.3.0 sphinx-copybutton==0.3.3 sphinx-panels==0.6.0 sphinx-rtd-theme==1.0.0 sphinxcontrib-applehelp==1.0.2 sphinxcontrib-bibtex==2.4.1 sphinxcontrib-devhelp==1.0.2 sphinxcontrib-htmlhelp==2.0.0 sphinxcontrib-jsmath==1.0.1 sphinxcontrib-mermaid==0.7.1 sphinxcontrib-plantuml==0.21 sphinxcontrib-programoutput==0.17 sphinxcontrib-qthelp==1.0.3 sphinxcontrib-serializinghtml==1.1.5 sphinxcontrib-spelling==7.2.1 statsmodels==0.13.0 tabulate==0.8.9 tenacity==8.0.1 terminado==0.12.1 testpath==0.5.0 textparser==0.23.0 textx==2.3.0 threadpoolctl==3.0.0 toml==0.10.2 tomli==1.2.2 toolz==0.11.2 tornado==6.1 traitlets==5.1.1 typing-extensions==3.10.0.2 uptime==3.0.1 urllib3==1.26.7 wcwidth==0.2.5 webencodings==0.5.1 widgetsnbextension==3.5.2 windows-curses==2.2.0 wrapt==1.13.3 wxpython==4.1.1 xlsxwriter==1.4.5`

foxBMS, 2024-07-02:
Closing this in favor of #49.

closed 2024-07-02 by foxBMS