foxBMS/foxbms-2 #50: foxBMS GUI log Parser not working

BhavaySen, 2024-07-10:
How to save and view trace files in log Parser?  because when I'm trying to view the file then it seems blank without any files.

BhavaySen, 2024-07-11:
Completed it by converting the required format. 

@foxBMS Team, it is my humble request, if you are providing any system with the name tag of open-source then provide complete information about it. 

Below points I observed that can become a kiss of death factor for a developer who trusts your system.

1. NXP PMICs are programmable and comes under NDA . [ not mentioned in the documentation]
2. What part of this project is actually open-source?  because ethernet PHY used in board is not open-source? right?

KnocheIsland, 2024-08-16:
The NXP parts can be programmed in situ using the NXP development platform and the files provided along with the hardware files. Recommend getting 

https://www.nxp.com/design/design-center/software/analog-expert-software-and-tools/sdk-analog-expert-drivers/fs84-fs85-fs66-safety-sbc-programming-boards:KITFS85SKTEVM

And acquiring the SBC-OTP-Config released with version 1.1.5
https://iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/hardware/release/master-v1.1.5_sbc-otp-config.zip

foxBMS, 2024-09-26:
> 1. NXP PMICs are programmable and comes under NDA . [ not mentioned in the documentation]
> 2. What part of this project is actually open-source?  because ethernet PHY used in board is not open-source? right?
> 3. The slave card MC33775A does not work in default configuration.

Dear @BhavaySen,

1. We will update the documentation in the next release.
2. Everthing that is part of this repository is open source. Due to foxBMS being an open source project you are free to implement a PHY driver on your own.
   However, you can have a look at our latest news post, we are planning to open source a PHY driver (https://foxbms.org/news/general/upcoming-foxbms-feature-highlights-in-early-2025/).
4. We have the default configuration working in our HIL setup. There might be some missconfiguration in your software or your hardware is not assembled correctly.

Best regards,
The foxBMS Team

closed 2024-09-26 by foxBMS