foxBMS/foxbms-2 #51: Missing pick and place csv file for slave ltc6813-1 v1.1.5

rexlee2, 2024-08-07:
Hi!

I am trying to fabricate foxBMS in order to customize it for energy storage (using sodium-ion). Upon trying to fabricate a slave board based on LTC6813, the pick and place CSV file seems to be missing. The file is available on the LTC6811 variant. May I request the PnP .csv file to be added? 

Also, it would be great if there is a quick guide on the PCB fabrication and which of the fabrication options (e.g. copper weight) that was assumed by foxBMS team. Thanks!

foxBMS, 2024-09-26:
Dear @rexlee2,

thanks for the report.

We will update the hardware release file to also include the pick and place csv file.

Best regards,
The foxBMS Team

foxBMS, 2024-12-20:
Dear @rexlee2 ,

The new release of the LTC LTC6813-1 v1.1.6 contains the pick and place csv file (see [iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/hardware/release/](https://iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/hardware/release/).

Best regards,
The foxBMS Team

closed 2024-12-20 by foxBMS