foxBMS/foxbms-2 #56: API documentation removed in V.1.7.0 release

BhavaySen, 2024-08-25:
Please share the other way to check the API documentation as the option to go to the API documentation page seems missing in this release.
@foxBMS

foxBMS, 2024-08-26:
Dear @BhavaySen,

The doxygen documentation is no longer linked in the general documenation.

Please have a look at the documenation https://docs.foxbms.org/software/build/build.html on how to build the doxygen documentation.

Best regards,
The foxBMS Team

BhavaySen, 2024-08-28:
Dear foxBMS team, 

I have build the docs by following the instructions but still not getting readable api documentation for easy access. Can you please guide on this?

Regards,
BVS

foxBMS, 2024-08-28:
Dear @BhavaySen,

You need to first build the doxygen documentation (e.g., `.\fox.ps1 waf build_doxygen_src`). Then you find the rendered documentation at `path/to/repo/build/doxygen_src/html/index.html`.

Best regards,
The foxBMS Team

BhavaySen, 2024-08-28:
Dear foxBMS Team,

Thanks for the quick response. The issue was because of the Doxygen and now it is resolved. 

Best Regards,
Bhavay Sen