foxBMS/foxbms-2 #61: Master crashes when configuring 2 slave strings

niktwo17, 2024-09-16:
Hi foxBMS-team,

when I set `#define BS_NR_OF_STRINGS (2u)` (in `battery_system_cfg.h`) the master instantly crashes, upon startup. Please see the picture below.
Tested with both releases 1.6.0 and 1.7.0. No changes made besides number of strings.

[image]

Thanks for your support,
Niklas

foxBMS, 2024-09-23:
Dear @niktwo17,

The BMS software works for multiple strings. You need to adapt the SPI configuration to be able to use multiple strings. If you do not adapt it, then the array containing the SPI configuration is initialized invalid and you will access invalid memory, therefore the BMS chrashes.

We will update the documentation's [how-to section](https://docs.foxbms.org/software/how-to/how-to.html) in the next release on how to use multiple strings.

Best regards,
The foxBMS Team

closed 2024-09-23 by foxBMS

niktwo17, 2024-09-23:
Thanks for the answer @foxBMS. However, I still think it makes sense to change code in a way where not only the first instance of a struct with configurable size is initialized - this will save others time in the future. As written in the PR, **I just outlined a possible solution - I trust you as maintainers to find a desirable way**. When developing and debugging with different BMS setups having to constantly change multiple files is a huge headache.