foxBMS/foxbms-2 #63: Configuration Needed for MC33775A Support

Anshul462046, 2024-09-25:
I am currently using the FoxBMS software, which is set up for the LTC slave card by default. I need assistance with configuring the software to work with the MC33775A.

1. What specific changes should I make in both hardware and software configurations to adapt the existing code for the MC33775A?
2. Are there particular settings or parameters in the configuration files that need adjustment?
3. Is there a checklist or ordered sequence of steps to ensure the application operates without errors or warnings?

Challenges: As I have configured at high level like in bms.json  but encountered some errors i am resolving it but is there any smart way or checklist while attempting to modify the code for the MC33775A. Any guidance or documentation regarding the integration of the MC33775A would be greatly appreciated.

foxBMS, 2024-09-26:
Dear @Anshul462046,

are you using the latest release of the NXP MC33557A-based foxBMS BMS-Slave (version 1.0.0, https://[iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/hardware/release/](https://iisb-foxbms.iisb.fraunhofer.de/foxbms/gen2/hardware/release/)).

This BMS-Slave works in combination with NSP MC33664-based BMS-Interface (version 1.0.2, available at the same link).

Which software version of foxBMS 2 are you using?

Best regards,
The foxBMS Team

anshulforwork, 2024-09-26:
Dear @foxBMS Team,

 Thank you for your response ,I am currently working with the NXP MC33557A-based foxBMS BMS-Slave (version 1.0.0) and the MC33664-based BMS-Interface (version 1.0.2). I am using software version 1.7.0 of foxBMS 2.

I have made the necessary changes in the bms.json file for high-level configuration. However, I noticed that most configuration files within the Application are set up for LTC-based slaves by default. I have a few questions regarding the software configurations:

Specific Changes for MC33775A:
What specific changes should I make in the software configurations to adapt the existing code for the MC33775A?

Settings and Parameters Adjustments:
Are there particular settings or parameters in the configuration files that need adjustment for the MC33557A?

Checklist for Application Setup:
Is there a checklist or an ordered sequence of steps I should follow to ensure that the application operates without errors or warnings?

Anshul462046, 2024-09-27:
Dear @foxBMS Team,

Thank you for your response ,I am currently working with the NXP MC33557A-based foxBMS BMS-Slave (version 1.0.0) and the MC33664-based BMS-Interface (version 1.0.2). I am using software version 1.7.0 of foxBMS 2.

I have made the necessary changes in the bms.json file for high-level configuration. However, I noticed that most configuration files within the Application are set up for LTC-based slaves by default. I have a few questions regarding the software configurations:

Specific Changes for MC33775A:
What specific changes should I make in the software configurations to adapt the existing code for the MC33775A?

Settings and Parameters Adjustments:
Are there particular settings or parameters in the configuration files that need adjustment for the MC33557A?

Checklist for Application Setup:
Is there a checklist or an ordered sequence of steps I should follow to ensure that the application operates without errors or warnings?

foxBMS, 2025-04-01:
Dear @Anshul462046,

we have identified the problem and have released a new version of this BMS-Slave that fixes it.
For details please see https://github.com/foxBMS/foxbms-2/issues/44#issuecomment-2768381751.

Best regards,
The foxBMS Team

closed 2025-04-01 by foxBMS