foxBMS/foxbms-2 #69: Without the buck-boost converter, how would you power the ISOMON, smart switches and interlock?

TaijiSugiyama1, 2025-01-09:
Greetings,
The production variant(which I believe is the version the foxbms team uses) does not have the buck-boost converter. Which provide the Vbb_12V to the ISOMON, smart switches and the interlock. Does this mean in the production variant, all these features does not work?

TaijiSugiyama1, 2025-01-09:
Yes, I just saw that there was a 0ohm Resistor to bypass the converter. Thank you for your swift reply

TaijiSugiyama1, 2025-01-10:
The SmartHigh-Side Power Switch TPS2H000BQPWPRQ1 is also missing in the production version. Does that mean the interlock functionality does not work? 
[image]

foxBMS, 2025-04-01:
Dear @TaijiSugiyama1,

The system is fully operational. It supports two distinct modes: static operation and dynamic/periodic operation. In the standard configuration, the interlock operates in static mode and is therefore powered by the 5V supply of the SBC.

Best regards,
The foxBMS Team

closed 2025-04-01 by foxBMS