foxBMS/foxbms-2 #71: Master Board : Watch-dog schematic issue

BitwiseGuru, 2025-03-05:
Hello @foxBMS team,

- **U6601 and R6607 Conflict:**

U6601 and R6607 cannot be present simultaneously in the same configuration. However, in the production variant schematic, both components are included:

[image]

- **WDI Input Pin Connection:**

The WDI input pin is connected to pin K4 on the MCU. According to the TMS570LC4357 HALCoGen configuration, this pin appears to be used as an edge input for HET2. However, the foxBMS configuration only enables HET1. Should this pin also be configured as an output, or am I missing a key detail?

Looking forward to your clarification.

BitwiseGuru, 2025-03-06:
After removing R6607 to enable an independent supply via the LDO and adjusting the N2HET2 configuration to keep using the K4 pin, which remains connected to the WDI input, the watchdog open window is properly catched.

foxBMS, 2025-03-10:
Dear @BitwiseGuru,
 
you are correct, only one supply path is neeed: you can either use the supply from the SBC, supply directly by bypassing the linear regulator, or remove R6607 and use the linear regulator. We will update the production variant along with the pin configuration in the upcoming release.
 
Best regards,
The foxBMS Team